solar3s/goregen #6: DataLog by the type of batteries

rubensben, 2017-06-27:
Hello World ! In order to differentiate the values of user/batteries types, would it be possible to : 
A - include in the the log file the name of the user + type of battery
or
B - name the log-file by the user and battery type.

AKA :
[User]
  Name = ""
[Battery]
  Type = ""
  Voltage = 0
  Brand = ""
  Model = ""

Goal is to differentiate users and batteries type for the BIG DATA interpretations

rkjdid, 2017-06-27:
on the roadmap yes.

rkjdid, 2017-06-27:
the idea is to include in each datalog the whole configuration used (including username, battery etc.), and then the actual data

rkjdid, 2017-06-27:
I'm leaning towards splitting each cycle into it's own file, named with this pattern "user-battery-chargeN.log" or something, merging them together afterwards shouldn't be too hard if needed

closed 2017-06-27 by rkjdid

rubensben, 2017-06-27:
ok looks solid ^^ thx !

Rubens Ben
Free Rubens <http://freerubens.com/>       Gif Arte <http://gifarte.com/>
 SolarSoundSystem <http://solarsoundsystem.org/>
   <http://freerubens.com/>        <http://gifarte.com/>
<http://solarsoundsystem.org/>
Mobile: +33(0) 695684355
Email: rubensben@gmail.com
<http://facebook.com/rubensphotos>  <http://twitter.com/freerubens>
<http://facebook.com/rubensphotos>  <http://instagram.com/suparub>