solar3s/goregen #8: CH341 défectueux

jparbel, 2017-07-26:
Dès les premiers tests, l'application goregen V0.3 ne fonctionne pas correctement, on est bien connecté, les boutons "Charge" "Discharge" etc ..  fonctionnent mais la tension indiquée est de 1400900 mV, il n'y a donc pas d'oscilloscope. En alimentant le CH341 seul sans le PCB, on lit la même valeur de tension.
Pour essai, j'ai donc remplacé le CH341 par un arduino UNO que j'avais sous la main, et là, tout fonctionne.
J'ai fait les essais sous WIN10 et Ubuntu 16.04.LTS avec les mêmes résultats.
J'ai seulement changé la valeur de CAN_REF à 4600 mV dans le programme firmware.ino pour qu'elle soit compatible avec mon UNO.
Donc méfiance avec les CH341 !
Est-il possible de m'envoyer un arduino NANO en remplacement du CH341 défectueux.
Jean Paul Arbel

rkjdid, 2017-07-26:
Cher Jean-Paul, il faut rentrer en contact avec quelqu'un de l'équipe, par exemple hello@regenbox.org - mais je doute que vous puissiez avoir un Nano officiel, un CH341 de remplacement par contre éventuellement. 

Je ferme l'issue qui n'a pas de rapport avec le projet goregen

closed 2017-07-26 by rkjdid

jparbel, 2017-07-26:
Merci pour votre réponse, mais pouvez-vous clarifier l'identité des
différentes équipes, leur rôle et leur contact.

Il y a trop d'informations dans différents endroits (parfois
contradictoires), j'avoue que je m'y perds, que je passe beaucoup de temps
à chercher la bonne info, et je pense ne pas être le seul dans ce cas.

Quels sont les liens entre goregen et regenbox ?

En finale, sur quel projet doit-on faire les tests officiels ?

Je pense qu'il faut un minimum de clarté pour pouvoir mener à bien ce
projet passionnant.

Jean Paul Arbel