solar3s/goregen #14: -dev flag

rkjdid, 2017-08-06:
should be passed down to connection watcher config, or it will try all available COM ports on potential connection loss.

https://github.com/solar3s/goregen/blob/master/main.go#L55-L63
https://github.com/solar3s/goregen/blob/master/regenbox/watcher.go#L17-L19
https://github.com/solar3s/goregen/blob/master/regenbox/watcher.go#L73

kepon85, 2017-10-26:
Je suppose que ça correspond à ça cet issu :
Quand je spécifie un device (via -dev /dev/ttyUSB1 par exemple) ça fonctionne puis dès qu'il y a un timeout il tente de se connecter mais ne considère plus l'argument, j'ai l'impression qu'il se connecte à /dev/ttyS0

Constaté sous Linux ARM et x64

rkjdid, 2017-10-26:
yep, je t'ai vu en parler sur slack je crois

kepon85, 2017-10-26:
C'est ça !

commit 5c8cbb21 (2020-02-14): main: -dev flag now locks underlying SerialPort fixes #14

also add the possibility to lock device in config.toml
closed 2020-02-14 by rkjdid