solar3s/goregen #25: Bad voltage reading with firmware 0.2 (goregen V0.4.1)

Chatmetaleux, 2020-07-04:
Hello,
I'm getting bad voltage reading with the new change in getVoltage() function.
[Here is a pic](http://chatmetaleux.be/screen_bad_voltage.png)
This is my firsts tests after building the regenbox.

With the first firmware of [regenbox git](https://github.com/solar3s/regenbox/blob/master/firmware/regenbox/regenbox.ino), v1.1, I can read a correct value of voltage.

So to make it works I replaced the getVoltage() function with the old one and it reports correct voltage values (checked with un voltmètre).

rkjdid, 2020-07-04:
bizarre, j'ai vraiment plus les mains là dessus mais ok pour le fix c cool

rkjdid, 2020-07-04:
d'après ce que je vois dans les release, y'a eu un patch entre v0.3 et v0.4, donc ptet jette un oeil au firmware.ino inclus dans la v0.3, si tu veux creuser davantage

Chatmetaleux, 2020-07-04:
oky je vais regarder à ça ;)

Chatmetaleux, 2020-08-02:
fixed in #27

Chatmetaleux, 2020-08-02:
fixed in #27

closed 2020-08-02 by Chatmetaleux