solar3s/goregen #26: goregen 0.4.1 log : firmware is out of date, please update...

Chatmetaleux, 2020-07-04:
Seams that the firmware does not give the version (no reply in the loop() function for firmware version message).
Do have I an old firmware version ? How to check ? I have the latest from this repository...

rkjdid, 2020-07-04:
so, as you may have noticed development stopped at some point, I'm really not sure the state of the art on this matters.
What I do know:
  - all firmwares are pretty much experimental, and can be tweaked, tested etc.
  - they live in the "firmware/" folder,
  - the "version" command, is not mandatory and does not alter behavior
  - the supposedly most stable should be firmware.ino, but you're welcome to try the other firmwares aswell, read them, combine them if you feel like it

Chatmetaleux, 2020-08-02:
fixed in 24a0c932bafd76b1c544a0af94afa2a0b06ca0d2

closed 2020-08-02 by Chatmetaleux